RethinkRobotics-opensource/rosnodejs #199: Only the last message is published

genkv, 2023-05-28:
```
function declearSubAndPubs(rosNode) {
  const nh = rosnodejs.nh;
  const customMessage = rosnodejs.require('my_package').msg.TransportRequestType;
  let pub = nh.advertise('/queue', customMessage);

  let sub = rosNode.subscribe('/chatter', customMessage,
        (data) => { // define callback execution
          rosnodejs.log.info('I heard: [' + data.type + ']');

          // send to pub
          pub.publish(new customMessage({type: 1}));
          pub.publish(new customMessage({type: 2}));
          pub.publish(new customMessage({type: 3}));
          pub.publish(new customMessage({type: 4}));
        }
);
```

Hi, have a question about the publisher. In my prototype code, where whenever a message is heard, I consequently publish some messages of the same type to a channel.

However, when I ```rostopic echo queue```, only the last message ```type: 4``` is printed. Kindly asking why. Should I wait a minimal number of milliseconds between each publish or do something else? thanks

chfritz, 2023-05-29:
The `publish` function takes a second argument that specifies a throttle in milliseconds. Try setting that to `-1` to override the default of `0`. You can also provide publishing options in the `advertise` call:
```
  let pub = nh.advertise('/queue', customMessage, {throttleMs: -1});
```
Please let us know if that worked so we can update the documentation.

genkv, 2023-05-30:
> The `publish` function takes a second argument that specifies a throttle in milliseconds. Try setting that to `-1` to override the default of `0`. You can also provide publishing options in the [...]

It works! Thank you so much

closed 2023-05-30 by genkv