RethinkRobotics-opensource/rosnodejs #204: Backticks in msg definition comments result in invalid JS syntax

chfritz, 2024-10-19:
In working with a user of [Transitive](https://github.com/transitiverobotics/transitive), I discovered that backticks in comments in `.msg` files result in auto-generated JS code that is invalid (the included backticks end the multi-line string we generated that itself uses backticks as delimiter). Unfortunately, there are even official ROS packages like [GridMap](https://docs.ros.org/en/jade/api/grid_map_msgs/html/msg/GridMap.html) that use backticks in their comments.

The good news is that it is easy to fix, and I already have a fix ready [here](https://github.com/chfritz/rosjs/commit/7fa86b859142f9766ca962f2e0e9a89b98c4f506).

@chris-smith I'm not certain how gennodejs works these days. Is it still using [this completely separate python implementation](https://github.com/RethinkRobotics-opensource/gennodejs/blob/kinetic-devel/src/gennodejs/generate.py) or does it use the message generation from this repo by now? If the former, then the above fixes would be easy to apply. If it's a completely separate implemetation in Python we would need to fix it there, too.

I'll create a PR for the node.js version for now.

closed 2024-10-21 by chris-smith

chris-smith, 2024-10-21:
`gennodejs` is still a separate python implementation, though we're using [`genmsg`](https://github.com/RethinkRobotics-opensource/gennodejs/blob/kinetic-devel/src/gennodejs/generate.py#L706) there so might not have to do anything?

chris-smith, 2024-10-21:
hmm actually it probably is just `gennodejs`? In `genpy` or `gencpp` they presumably aren't using backticks so not escaping them is fine.

chfritz, 2024-10-21:
In that case we can probably just sanitize the result we get from `genmsg.compute_full_text`?

chris-smith, 2024-10-22:
sounds right to me

PR #205 (2024-10-21): Fix backticks in message definitions, throw better error on missing dependencies.
Fixes #204 -- at least for this repo.

- [Message gen: escape backticks in comments](https://github.com/RethinkRobotics-opensource/rosnodejs/commit/21c3e46d15808f29ad6b8cd2bca31c9cc9ff3e6c) 
  Until now, backticks ("`") in comments in message definitions will lead to an exception when the generated message is loaded, because the generated JS code is syntactically invalid.

- [Message generation: indicate name of missing dependencies](https://github.com/RethinkRobotics-opensource/rosnodejs/commit/9966bcabd7adf1949 [... cut]